cprezzi/grbl-LPC #1: Port Spindle Direction Pin

Rowbotronics, 2017-10-25:
We need the spindle Direction output to run our laser properly. Was there a technical reason that this output was not ported? We're looking at adding this functionality, but wondering if there are any known issues.

cprezzi, 2017-10-25:
I don't know why it was not ported, as this is just a fork of the original port at https://github.com/gnea/grbl-LPC from @tbfleming.
 
Wouldn't it be possible to use a manual switch (or door switch) instead of the spindle direction pin? The laser would still not fire until the spindle PWM is > 0. This is how most K40 settings work.

tbfleming, 2017-10-25:
I didn't port it because lasers don't typically need it. What does spindle direction do? Reverse photon flow?

tbfleming, 2017-10-25:
I forgot that it's a shared pin on the 328. If you need an enable pin, then create it as one, not a weird shared enable/direction pin like grbl-328 has.

Rowbotronics, 2017-10-25:
@tbfleming In Laser Mode the spindle direction pin acts as an enable pin; on when the laser is firing, off when the laser is off. Our power supply has PWM and enable inputs, which were both in use with the previous control board.

@cprezzi That's how we have it set up currently, and it works fine for cutting and very slow engraving. The problem seems to be that the Laser output lags the PWM input by a noticeable amount, which leads to a distinct comb-like pattern on the edges of quicker engraves. The previous control board would hold the PWM at a constant level and turn the Laser on and off by toggling the enable pin.

As a little more background, we have an 80-watt 1200x900 Laser Cutter, so we're probably shooting for higher speeds than most K40 users. This may not get us all the way back to the speeds of the original (recently deceased) control board, but the trade-off of much more usable software is worth it for us.

Rowbotronics, 2017-10-25:
@tbfleming Yeah, looking at the code it doesn't seem like it would be too hard to port, so I was wondering if there was a reason that I wasn't seeing that it didn't get ported. If the reason was "not worth the time because it wasn't necessary" then that is perfect. We'll take a crack at it.

cprezzi, 2017-10-25:
I do run picture engraving on my K40 with grbl-LPC up to 300mm/s and don't get any visible activation delay. Are you sure it's not backlash of your machine that you see?

Did you try the "Over Scan" feature in the "Laser Raster" operation? This would cause the line to be longer than the picture width, so the LPS has time to start.

Rowbotronics, 2017-10-25:
Yes, we've been using an "Over Scan" setting of up to 30mm when testing raster engraves. I very much doubt that the issues are related to backlash; this machine was running and engraving just fine a week ago when the control board failed and the previous software had no backlash compensation settings (at least none that I could see, the interface was barely english). Also, using the "Laser Fill" operation to engrave vector files leads to perfectly filled shapes, probably because the Laser doesn't have to turn on and off constantly. We've also tested using Constant Laser Power Mode with the same results.

If you look at the edges of the engraves you can see engrave path getting wider and deeper as the Laser power ramps up.  I'll post a picture when I get back to the machine today.

Rowbotronics, 2017-10-26:
This one was run at 300mm/s with no overscan.  You can see at the far left and right edges there is no gap before the start of the engrave line, but on the interior pieces when the laser has to turn on and off. It seems like the longer the laser is off, the longer it takes to turn back on when the PWM starts up again.

Going to try to get the enable pin going in the next few days.

[image]

cprezzi, 2017-10-26:
Looks like your laser tube and LPS combo needs longer to start than usual. You could try to define the min power > 0% in the operation to hold the tube activated with very low power.

Rowbotronics, 2017-10-26:
Yeah, the laser starts firing at 13%, so we have PWM min value set to 13% and PWM off value set to 12%, but that hasn't seemed to help any.

Rowbotronics, 2017-10-30:
I got the enable pin code ported and working. However, it doesn't behave like I thought it would. The enable does not toggle off during G0 moves, only when the spindle is disabled completely with M5, which also stops the head. This makes it unusable for engraving. Looks like I'll have to modify the way GRBL works in laser mode.

cprezzi, 2017-10-31:
There are many params involved in the process. Delay of tube activation is one, which you can't solve with software only, but the effect should decrease with lower speeds.

The PWM frequency can eventually also cause problems. Lets say you use 2000Hz, which is just one pulse every 0.5ms. At 300mm/s, this is 0.15mm! The question is, how many PWM pulses the LPS needs to establish a stable laser beam.

Rowbotronics, 2017-11-01:
Right. We're running it at 20kHz, but we don't know how long it takes our LPS to stabilize. I assume it has some kind of low-pass filter on the PWM input to convert to an analog voltage which will certainly introduce some delay. This is another reason why I think that feeding it a constant PWM and just toggling the enable will be much faster at switching the supply on and off.

cprezzi, 2017-11-01:
You should check Don's Blog for more Details on LPS and reaction time: http://donsthings.blogspot.ch/search/label/K40%20PWM%20Control

Rowbotronics, 2017-11-02:
Thanks for the link, there's a lot of interesting info there.

I tested a few different connection configurations yesterday; the best results were when I ran the PWM output at 100% into the enable input, and set the laser power with a potentiometer. When I ran a variable PWM into the enable input I had to decrease the frequency to about 3khz, and it was very noisy. I think I'm on the right track trying to get the enable to set G0 moves to toggle the enable off and leave the PWM on.

cprezzi, 2018-04-14:
I closing this old issue due to not beeing a general firmware issue.

closed 2018-04-14 by cprezzi