cprezzi/grbl-LPC #2: Not receiving PWM on pin 2.5 on Cohesion3d Mini board

PeterGrace, 2017-12-16:
Hello Claudio,

I have setup my C3d mini board based on the documentation here: https://github.com/LaserWeb/LaserWeb4/wiki/2.2-Initial-Configuration---GRBL-LPC-1.1

I have downloaded the latest firmware on your release page.

I am not receiving a PWM signal on P2.5 on the C3D mini board.  I've got an oscilloscope hooked up to the pin and the pin is staying a constant 4.8V.

Given the setup outlined in your docs, what diagnostic steps might I be able to employ to troubleshoot the issue?

P2.5 was properly doing PWM for me with Smoothie, so I believe that this is either a misconfiguration on my side, or a firmware issue.

Here's my settings.  The only thing I've actively changed is $10 where I set it to 1 so that bCNC would communicate with the board appropriately.

```$0=10
$1=255
$2=0
$3=3
$4=0
$5=1
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=1
$21=0
$22=1
$23=1
$24=50.000
$25=6000.000
$26=250
$27=2.000
$30=1000
$31=0
$32=1
$33=5000.000
$34=0.000
$35=1.000
$36=100.000
$100=160.000
$101=160.000
$102=160.000
$103=160.000
$110=24000.000
$111=24000.000
$112=24000.000
$113=24000.000
$120=2500.000
$121=2500.000
$122=2500.000
$123=2500.000
$130=300.000
$131=200.000
$132=50.000
$133=100.000
$140=0.600
$141=0.600
$142=0.000
$143=0.000
```

PeterGrace, 2017-12-16:
as a follow up, the firmware on Cohesion3d's site works: https://cohesion3d.freshdesk.com/support/solutions/articles/5000740838-grbl-lpc-for-cohesion3d-mini

cprezzi, 2017-12-17:
Did you download/install the correct version for the C3d mini? 
How do you measure P2.5? The real MCU pin before the Mosfet, or something after the Mosfet?

PeterGrace, 2017-12-18:
I downloaded c3d_mini_firmware.bin which has an md5sum of 3bc719e5cb3f9fed7be3a8090d2472d6.

I measured P2.5 at the screw terminal on the board, which is after the mosfet.  I'm not sure if there's an exposed pin for 2.5 before the mosfet, but I'll take a look around and see if I can duplicate the results.

cprezzi, 2017-12-18:
The pin 2.5 is not exposed, but you can measure it at the Mosfet pins on the board.

The N-Channel Mosfet does only switch the ground side (like an open collector transistor), so you have to connect a source to the Mosfet Power In to measure the Mosfet output (between the two pins).

PeterGrace, 2017-12-19:
The firmware from the Cohesion3d website does produce a proper PWM signal on the P2.5 screw terminal; there must be some difference between the two firmwares that would cause that.  If you want me to do something specific to test, I will be happy to, but I'm otherwise pretty well set with the C3D firmware instead.

cprezzi, 2017-12-19:
The firmware on the Cohesion3d page is based on mine, just a little earlyer state. I just added support for a 4th axis, but no other relevant changes. So you can just take the C3d version.
I guess there is a configuration error in my C3d mini firmware version. It could be that pin 2.4 is used for PWM-out, instead of 2.5. I will check that.

cprezzi, 2018-04-14:
The default for C3d Mini PWM Pin is P2.5. If P2.4 is needed, then activate #define SPINDLE_PWM_PIN_2_4 in config.h.

closed 2018-04-14 by cprezzi