cprezzi/grbl-LPC #5: Add dockerfile

jorgerobles, 2018-04-29:
Hi Claudio.. I've found this: https://github.com/ryankurte/docker-arm-embedded and runs perfectly!

cprezzi, 2018-04-30:
What is this for? I don't know anything about docker ;)

jorgerobles, 2018-04-30:
Docker is a containerzation engine (something similar to a virtual machine) 
Just adding the dockerfile to this repository makes possible to anyone having docker installed to run a command to start the virtual machine, download all the dependencies and start a command line inside de virtual machine

Once inside, you only have to run `make` :D

cprezzi, 2018-04-30:
Ok, why not. Can you add it and make a PR?

jorgerobles, 2018-04-30:
I suppose, it's MIT licensing, does not interfere isn't it?

cprezzi, 2018-04-30:
I don't think so. Grbl-LPC runs under GPLv3 which can be combined with MIT licensing.
(see: https://stackoverflow.com/questions/3902754/mit-vs-gpl-license)

cprezzi, 2018-04-30:
Ok, merged. Thank you.

closed 2018-04-30 by cprezzi

jorgerobles, 2018-04-30:
:)