cprezzi/grbl-LPC #8: M8/M9 coolant not working

jimfong1, 2018-05-15:
Hi Claudio.
Testing out your October 8th release version of c3d_remix_firmware.bin 

I can toggle p2.7 with M7 and M9. I can see the pin change on the oscilloscope. No problems there.

M8 will set p2.6 high but M9 will not set it back low again.  Took a look at coolant_control.c but all looks correct.  When you get a chance, can you take a look.

Thanks,

Jim

cprezzi, 2018-05-29:
You are right. I found the same problem. The source code is looking corect to me too. 
I need to investigate a bit more...

cprezzi, 2018-06-13:
I'm sorry, I did not yet have time to look into it.

jgilbertfpv, 2018-12-16:
Is this working? I was curious about setting up my air assist on p2.7 used to trigger a relay to turn on the compressor

gerrylenz, 2019-07-15:
take a look in
https://github.com/cprezzi/grbl-LPC/issues/24

jimfong1, 2019-07-16:
Thanks that worked.

closed 2019-09-20 by cprezzi