cprezzi/grbl-LPC #9: PWM pinout at 24 volts?

PhysicsAnonymous, 2018-08-14:
I got everything working my MKS board expect the PWM for laser power. this is a rebuilt k40 machine (nothing left of the original except the power supply and the tube). Initially I tested the voltage on the plus and minus pins on the mosfet connector (pin2.5) and was getting between 20 and 24 volts during cuts. I am guessing this is because i have the MKS hooked up to a 24 volt supply. I tuned the power level of the laser down to 14 percent as the max and got a max of 5 volt out put during a cut, but when i hooked these pins up to the pot pins on the laser supply, it went back up to 24 volts, even at idle.

the documentation isn't very clear on how to hook the PWM pins to the power supply for the k40, so i am probably making a mistake. any help you can provide would be appreciated.

PhysicsAnonymous, 2018-08-14:
I should also mention have a type A power supply also known as k+, and i got the whole thing working fine on an arduino 328p, but the chip wasn't fast enough for the speed the machine is capable of

cprezzi, 2018-08-15:
The usual K40 power supply has a L pin (on the rightmost connector) and a IN pin (on the pot connector). The IN pin is only used by the pot to set the maximum laser power (mA). 

The L pin activates the laser when pulled to GND and should be connected to the minus pin of the P2.5 mosfet. The positive pin of the mosfet is NOT used. GND of the power supply and the MKS board should be connected. The L pin is pulled up in the power supply (voltage does not matter), it just needs to be pulled down to GND to activate the laser, which exactly happens by the mosfet (P2.5 PWM).

Laser Switch and Test Switch can stay as the are originally,

PhysicsAnonymous, 2018-08-15:
Thanks cprezzi, that helps a lot. I ended up wiring it up through an opto-isolator and a resister from the input pins to the mossfet, it works, but i will switch it to the way you have described just for simplicity.

elmo-poog, 2018-09-04:
Good afternoon guys 

If anyone can point me in the right direction I really need the help. I am converting a 900x800 laser cutter from the stock board to an Mks Sbase v1.3 running GRBL-LPC port by cprezzi using laserweb4.0. My HV power supply is an ZYE MYJG50, I have managed to get everything running apart from the PWM from the negative p2.5 pin from the MOSFET.  I was able to located the L pin on the MYJG50W and GND with no problem. When I manually tap the wires together or hit the fire button on the front of the unit it activates leaving me a nice little hole in my test card. 

But when I attach it as described above, with black cable going to ground on the MKS and the red cable going to the negative pin of p2.5 and hit fire in the movement tab of laserweb4 nothing happens. When i probe the attached GND and L (p.2.5) there is a value of about 24v in an un-active state and when I hit fire on the laserweb4 no change it stays at 24v. I assume if I have all the powers at 100% (both on the pot and on the laserweb settings) this should pull the voltage to zero for a moment?

Does anyone know where I am going wrong? 
Thanks for the help!

cprezzi, 2018-09-04:
Did you configure the "Laser Test" feature in settings/gcde? By default it is deactivated for security reason.

elmo-poog, 2018-09-04:
Cprezzi, thank you, your a legend! It was the value for the max pwm, I had left it at 1 but for GRBL I needed to use 1000.

cprezzi, 2018-09-05:
@elmo-poog I'm happy that I could help.

closed 2018-09-05 by cprezzi