cprezzi/grbl-LPC #12: Dual-Y stepper setup

jgilbertfpv, 2018-12-26:
Is there any way to use one of the free plugs for E0 or E1 on a mks/smoothieboard as a second pinout for a dual-y axis configuration? I'm currently hooking two external drivers up in parallel but they're just cheap tb6600's and the onboard stepper drivers seem to be working alot better so i'd like to be able to do away with the externals and run everything directly on the mks board.

cprezzi, 2018-12-26:
Did you try to just bridge the step and dir pins of the y axis to one of the spare axis?

jgilbertfpv, 2018-12-26:
So just run my jumper from the step and dir pins on the external driver plugs on the board underneath the original drivers?

jgilbertfpv, 2018-12-27:
i just tried running a jumper across from the step/dir pins on y over to e1 step/dir pins and it didn't turn on the stepper plugged into e1

cprezzi, 2018-12-28:
You also need to bridge the enable pin to activate the driver.

cprezzi, 2018-12-28:
You could try it with E1, as E0 is used by this firmware for A asis.
Unfortunately grbl doesn't support dual axes in software, sorry.

onezero1010101, 2018-12-28:
@jgilbertfpv Please excuse me if I am out of place here, or am mistaken in the question.  But my machine is an MPCNC, and it has dual steppers for x and y axis, but we only use one stepper driver on the board to control each.  Here is a diagram of how we have them wired.  As well as a link to the wiring instructions.  Maybe it will help with what your trying to do?

https://www.v1engineering.com/wp-content/uploads/2018/10/513px-Series_Connected_Stepper_Motor_Wiring_Diagram-171x200.jpg

https://www.v1engineering.com/assembly/wiring-the-steppers/

jgilbertfpv, 2018-12-29:
@cprezzi jumping it to E1 seems to do the trick, thanks for the help

closed 2018-12-29 by jgilbertfpv