cprezzi/grbl-LPC #13: Disabled Current Control?

onezero1010101, 2018-12-26:
I am currently using Smoothieware on Smoothieboard, and am not sure if this is the same thing. But in my config I am setting the stepper motor current.  Does your fork disable the ability to set this in the config?  If that's the case, what current will be going to the steppers.  Thanks, and sorry to have to open an issue to ask a question.

cprezzi, 2018-12-28:
In GRBL, settings either has to be compiled into the bin (defaults) or made by $key=val commands over the USBserial connection. A config file on the SD card will not be used.
The Smoothieboard version uses the software current settings that can be configured with $140-143.

onezero1010101, 2018-12-28:
Thank you very much, I got the mks firmware loaded and initial testing looks to be working as expected.  Great work!

closed 2018-12-28 by onezero1010101