cprezzi/grbl-LPC #15: Smoothieboard, GRBL-LPC and LaserWeb

Kgv1, 2019-01-22:
I’m new to laser engraving but really enjoy the project I currently working on.

I make a CO2-laser cutter and planning on using GRBL-LPC firmware on the smoothieboard and use LaserWeb to send the G-code. I have some problem with the firmware, and I don’t know if I saved the right files on the SD-card. Is it correct to download the ZIP-file, save those files on a clean SD-card and reboot the smoothieboard?

I would really appreciate any help. Thanks.

cprezzi, 2019-01-22:
Just read the description on the release page.

> Download your board specific firmware (.bin) above to the SD card and rename it to firmware.bin, then reboot!

Kgv1, 2019-01-22:
I do not need the source code at all?

cprezzi, 2019-01-23:
You only need the bin file to flash the firmware for a Smoothieboard. All my compiles are made with specific board/machine settings. You only need the source files when you need to compile your own firmware for different boards/machines. In grbl, most settings (like axis dimention, stepps/mm, acceleration...) are made with $xxx=yyy commands over serial console on runtime. See https://laserweb.yurl.ch/documentation/initial-configuration/65-grbl-lpc-1-1e for details.

I suggest to save the Smoothie SD card for backup and use another empty card for grbl-LPC.

On each boot up the bootloader on the Smoothieboard will check the SD card for a file called "firmware.bin". If this file is found, it is flashed to the boards flash memory and the file is renamed (so it will not be used again).

After flashing, grbl-LPC is not using the SD anymore.

closed 2019-01-23 by cprezzi

Kgv1, 2019-01-23:
Great
Thank you so much for supplementary answers, I will try this out.