cprezzi/grbl-LPC #19: Z Probe connection on MKS SBase board

Dan018, 2019-04-07:
I'm reusing a spare MKS SBase board on a CNC router I'm building. I've set up the motion fine, so next I want to wire in a Z probe with a copper plate and crocodile clip.

On the Uno CNC shield you connect between pin A5 (SCL on the  shield) and ground and then run a G38.2 to get the distance moved.

cpu_map.h only shows PROBE_BIT 5, with the pin etc "Not Used". Is the probe feature available, and if so which pin should I connect to?

Thanks.

cprezzi, 2019-04-08:
Grbl-LPC was mainly made to get higher raser laser performance out of a smoothieboard (or compatible). CNC Milling was not on the target list and therefore probing is not yet supported.

closed 2019-04-08 by cprezzi

tscofield, 2019-07-26:
@Dan018 , if you are still in search of this feature, try building a firmware file from the latest code.  You will need to connect the probe pin to P1.27 and the other end to ground, This should be the Y+ limit switch on your SBASE board, which is not currently used