cprezzi/grbl-LPC #20: Booting mks sbase v1.3

nicooler, 2019-04-18:
Hi there,

I tried starting using the mks_sbase_p1_23_firmware.bin but it seems to not boot.#
Then I tried a view other fimwares and in grbl-LPC all of them arent working anymore.
Even build it myself, still not working and I know it was working before. I tested 2 Boards now, but no luck.
When connecting with Laserweb this happens:
[image]

I tried the smoothieware master firmware and that one seems to work. I can connect with the mks base. 

Do you have any idea?

Thanks Nico

pmth, 2019-06-16:
I  am getting `no supported firmware detected as well` for this board. Tried to add `RESET_ON_CONNECT=1` to plist.info (I am using mac) as well. The firmware seems to be installed, I noticed `firmware.cur`. Not sure if anyone can help, I am using this board:
https://www.amazon.com/BIQU-Smoothieboard-Controller-Source-Smoothieware-MKS-Upgraded/dp/B07PQ7LXK8/ref=sr_1_2?keywords=MKS+SGEN&qid=1560661489&s=industrial&sr=1-2

cprezzi, 2019-06-17:
Please go to https://forum.makerforums.info/c/laserweb-cncweb for community support.

Nonstopich, 2019-06-29:
> I am getting `no supported firmware detectedl`. Tried to add `RESET_ON_CONNECT=1` to plist.info (I am using mac) as well. The firmware seems to be installed, I noticed `firmware.cur`. Not sure if anyone can help, I am using this board:
> https://www.amazon.com/BIQU-Smoothieboard-Controller-Source-Smoothieware-MKS-Upgraded/dp/B07PQ7LXK8/ref=sr_1_2?keywords=MKS+SGEN&qid=1560661489&s=industrial&sr=1-2

Hello. I have same problem. The same board. Did you resolve it?

theomajigga, 2019-07-23:
Your accel/motion rates are too high and the machine is stalling causing a reset, is one of my guesses.

I use the board with no issues. Please try to do it with no steppers connected and see if you send move commands.

cprezzi, 2019-07-23:
@theomajigga I think you are wrong. This has nothing to do with firmware settings.

@Nonstopich The SBASE firmware doe not work on the SGen board, because there is no I2C stepper current chip on the SGen! You would need to compile your own version with separate settings for SGen.

theomajigga, 2019-07-23:
Misread it.

closed 2019-09-20 by cprezzi