cprezzi/grbl-LPC #22: MKS-SBase and PWM questions

FlailAway, 2019-05-23:
Is this still being monitored?

I have a very old Synrad 25w laser that is working fine, but the supporting electronics are toast. I have an MKS-SBASE board from a printer and have installed grbl-LPC and can connect just fine with LaserWeb. All good.

The laser has it's own inbuilt HF etc and just needs a PWM to start, control and stop. It needs a tickle pulse. From the manual:
Control (CTRL) Input
Logic Low (Off State) … 0.0 V to +0.5 VDC; 0.0 V nominal
Logic High (On State)… +3.5 V to +10.0 VDC; +5.0 V nominal
Maximum Current Load… 6 mA @ +5.0 VDC
Tickle Pulse Signal
Tickle Frequency 5 … 5 kHz
Pulse Length… 1.0 μs ± 0.2 μs
Pulse Rise/Fall Time… < 100 ns between +0.5 V to +3.5 VDC
PWM Command Input Signal
PWM Frequency 6 … DC to 20 kHz
PWM Duty Cycle … 0% to 100%

I will be using a 12vdc power supply for the MKS, but do NOT want to fry the Synrad inner-electronics (10vdc tollerant) so I need some guidance on connecting the PWM. I read about connecting the Negative side of 1.25, but not sure that is will go over well with the Synrad.

Should I be using the TTL 1.23 build? If so, I still need some guidance.

Could someone please help me get this going?

Thanks

cprezzi, 2019-05-24:
You need the special version with PWM on pin 1.23 (0..5V TTL PWM).

bsmith07, 2019-05-30:
cprezzi,

I'm in a similar situation.  Sbase v1.3 controller with the 1.23 version of GRBL-LPC installed.  I'm using the laser below

https://www.banggood.com/450nm-5W-Laser-Engraving-Module-Blue-Light-With-TTL-Modulation-p-1337358.html?rmmds=myorder&cur_warehouse=USA

There isn't a traditional power supply on these modules. I have a pwm + and -.  I've used pin 1.23 with the ground pin two pins over, but I can't get it to turn on and off with laserweb 4.  How do you configure a 2 pin pwm/ttl laser diode driver?  I've spent quite some time reading into the config.h and cpu_mapping.h files, but all I find seems to revolve around a single pin signal wire back to the PSU.  Any help would be greatly appreciated.

Brian

cprezzi, 2019-05-31:
Connecting only one pin will not work. It's allways needed that the controller board and diode driver use the same GND.

Please use our forum @ https://forum.makerforums.info/c/laserweb-cncweb for further assistance!

bsmith07, 2019-05-31:
cprezzi,

I created a topic here to discuss this issue, although I thought this had more to do with the board and firmware.  If the below topic is out of place, my apologies.
https://forum.makerforums.info/t/sbase-v1-3-grbl-lpc-pwm-issues-with-diode-module-lasers/77707

Thanks,

Brian

closed 2019-06-14 by cprezzi