cprezzi/grbl-LPC #24: [solved] M9 don't stop M8 Pin

gerrylenz, 2019-07-15:
M8 will set pin high or low but M9 will not set it back again.

original code in coolant_control.c
```javascript
void coolant_stop()
{
  #ifdef INVERT_COOLANT_FLOOD_PIN
    COOLANT_FLOOD_PORT |= (1 << COOLANT_FLOOD_BIT);
  #else
    COOLANT_FLOOD_PORT &= ~(1 << COOLANT_FLOOD_BIT);
  #endif
  #ifdef ENABLE_M7
    #ifdef INVERT_COOLANT_MIST_PIN
      COOLANT_MIST_PORT |= (1 << COOLANT_MIST_BIT);
    #else
      COOLANT_MIST_PORT &= ~(1 << COOLANT_MIST_BIT);
    #endif
  #endif
}
```
change to

```javascript
void coolant_stop()
{
  #ifdef INVERT_COOLANT_FLOOD_PIN
    COOLANT_FLOOD_PORT |= (1 << COOLANT_FLOOD_BIT);
    delay_ms(1);
  #else
    COOLANT_FLOOD_PORT &= ~(1 << COOLANT_FLOOD_BIT);
    delay_ms(1);
  #endif

  #ifdef ENABLE_M7
    #ifdef INVERT_COOLANT_MIST_PIN
      COOLANT_MIST_PORT |= (1 << COOLANT_MIST_BIT);
    delay_ms(1);
    #else
      COOLANT_MIST_PORT &= ~(1 << COOLANT_MIST_BIT);
    delay_ms(1);
    #endif
  #endif
}
```
In block #ifdef ENABLE_M7 i write too delays im not sure is necessary. I think this is a bug in toolchain.

Regards
Gerry

cprezzi, 2019-07-15:
Interesting. It doesn't make sense that a delay has any effect on this.

gerrylenz, 2019-07-15:
I think the compilers optimization is wrong.

cprezzi, 2019-07-15:
That might be. I will make some tests...

cprezzi, 2019-11-19:
I have added this workaround to the code, which seems to solve the issue.

closed 2019-11-19 by cprezzi