cprezzi/grbl-LPC #26: MKS SGEN board

TFMvdBroek, 2019-07-21:
Would this board work with GRBL-LPC? It is a bit cheaper than the MKS SBASE and seems to be a new version.

TFMvdBroek, 2019-07-21:
Pinout seems to be the same as they include the MKS SBASE pin file https://github.com/makerbase-mks/MKS-SBASE/blob/master/Firmware/Marlin-bugfix-2.0.x/Marlin/src/pins/pins_MKS_SGEN.h

cprezzi, 2019-07-22:
It will not work with the SBASE firmware image as there is no I2C Stepper Current Chip on the SGen.
You would have to make separate settings in cpu_map.h and compile your own version.

TFMvdBroek, 2019-07-22:
Guess I'll be using an SBASE then.
I can't seem to find a decent wiring guide (for K40). Do you perhaps have one?

cprezzi, 2019-07-23:
That's a good decision.
Leaving the pot connected to 5V, IN and GND and connect L to negative of P2.5 for PWM is the right decision..

closed 2019-09-20 by cprezzi