cprezzi/grbl-LPC #30: Re-ARM + Ramps 1.6 Motors Switched

krizo96, 2019-09-14:
I just flashed the latest version of grbl-LPC (Smoothie) to my new Re-ARM board with Ramps 1.6 and something weird is going on. I assume it's probably user error, but to move the X-axis I actually need to plug the motor into the Y, and to move the Y-axis it needs to be plugged into E0, while the others seem to do nothing at all.

cprezzi, 2019-09-16:
The Smoothieboard version of grbl-LPC is not compatible with Re-ARM boards! You need at least to correct the pin mapping in cpu_map.h and compile a special version.

krizo96, 2019-09-16:
I figured that, so I did just as you said last night and it's working fine now.
Only other thing I'm wondering is if it's possible to use all 5 drivers with grbl-LPC?

cprezzi, 2019-09-16:
The actual code only supports 4 axes yet, but should be extedable to 6 axes with not too much effort.

I would be happy to include your Re-ARM config into my branch. Could you send me your changes or make a pull request?

krizo96, 2019-09-17:
I'm not sure if  I have the Z pins are correct, as I only needed two axis.
While I'm at it, I can't seem to find the pwm pin on the Re-ARM/Ramps where I'd connect a laser or a spindle. What would be the easiest way to find it?

cprezzi, 2019-09-17:
You could use one of the servo outputs (if you need logic level pwm) or one of the 3 mosfets (for open drain PWM like on CO2 laser psu).

krizo96, 2019-09-17:
I'm sorry, I'm afraid you're gonna have to dumb it down a little for me here, I'm don't have much knowledge in this area and am just trying to make do.
Would I also change that in cpu_map.h? Because I'm not finding where the pwm is defined in there.
I tried figuring out which pin is acting as the lasers pwm manually by running a gcode and checking individual pins with a multimeter, but I didn't manage to find it. Probably was a dumb idea.

cprezzi, 2019-09-17:
If you started with the smoothieware config, then the PWM pin is defined as 2.5, which is the big mosfet (heatbed) on the RAMPS board. This Mosfet is pulling the line to GND when a PWM signal is active, which is good for laser power supplys with TL or L input.

What exactly do you like to connect to the PMW output?

krizo96, 2019-09-17:
I'm looking to connect a diode laser module with pwm/ttl at this point.

krizo96, 2019-09-19:
> You could use servo output 4 (p1.18) if you need logic level pwm @ 5V, or one of the mosfets (p2.4 or p2.5) for open drain PWM like on CO2 laser psu.
> 
> (P2.5 is the default PWM output also on the Smoothiebiard version)

Alright, I think I figured out that the P2.5 is the D9 on Ramps, but I'm not getting a pwm signal, it seems constant regardless of the image I'm trying to engrave and I end up with a burned blob.

cprezzi, 2019-09-19:
Did you test that at the bed connector or the digital pin?

krizo96, 2019-09-19:
I did both, but something else was the problem (power wiring) and once I fixed that it's working at least somewhat.
Only problem now is that it seems to be firing very weakly, even at 100% it's nowhere near as strong as a 5W laser should be (and was when I was using an Arudino Uno).
I tried adjusting the frequency $33 and the lower I set it the more powerful it is, but the image fidelity also goes down. And with anything higher than around 3000 it doesn't even leave a mark. And for reference in my laser module specs it says "TTL/PWM Control: 20kHz @5V".

cprezzi, 2019-09-19:
Check that the max. PWM value in GRBL and LaserWeb ist set equaly. Default in Grbl ($30) is 1000, so the [PWM MAX S VALE] in LaserWeb>settings>gcode should also be 1000. In the cam operation, the power setting is %, so 100 is max.

krizo96, 2019-09-19:
> Check that the max. PWM value in GRBL and your GCode is set equaly. Default in Grbl ($30) is 1000. If you use LaserWeb, set the [PWM MAX S VALE] in settings>gcode also to 1000 (in the cam operation, the power setting is %, so 100 for max.).

Well, I feel like a complete idiot now. I was so sure I already checked that, but it looks like that was the problem. Max was set to 1000 in grbl and 255 in the gcode.
Thanks for all the help, if I manage to correctly set all the pins I'll send you the changes so you can add Re-ARM for others who may need it.

closed 2019-09-20 by cprezzi

ej0rge, 2019-11-08:
Has this been merged? I have a spare ReARM/RAMPS-1.6 combo and just got my K40.

cprezzi, 2019-11-13:
@ej0rge No, there is still no default config for Re-ARM, so you need to add your own profile to cpu_map.h