cprezzi/grbl-LPC #31: MKS SGEN-L possible?

ej0rge, 2019-11-08:
I ordered an MKS SGEN-L for my K40 before realizing it isn't just an SGEN running 20mhz slower for less than half the price. I do understand that i would have had to cook up an SGEN configuration without support for the digipots on the SBASE regardless. I can probably handle that. 

For some fool reason, it's a wholly new design that appears to be inspired by the SKR. 

I saw the closed issue where it's stated that the SKR is not compatible because devices are on different ports, but i don't understand the lpc176x architecture well enough to understand what that means and thus how to tell, say from the marlin pins file, whether the SGEN-L is a candidate for grbl-lpc. 

It looks like Re-ARM is probably workable and i turn out to have a spare Re-ARM set, so it's no big deal if i spent $18 on a 3d printer controller that i don't have a use for.

cprezzi, 2019-11-11:
Neither SGEN nor SGEN-L are compatible with grbl-LPC, because they have incompatible pin mappings for the stepper drivers! 
Grbl needs to have all step pins on one port, and all directions pins on one port, which is not the case on the SKR or SGEN boards. This is a limitation of the grbl code structure.

closed 2019-11-11 by cprezzi

ej0rge, 2019-11-11:
The SGEN has the exact same pin mapping as the SBASE which seems to be supported? It just doesn't have the integrated stepper drivers, and has an lpc1769 instead of lpc1768.

cprezzi, 2019-11-12:
@ej0rge You are wrong. 
The SGEN has X step on **P2**.2, Y step on **P0**.19, Z step on **P0**.22 (different ports!)
The SBASE has X step on **P2**.0 and Y step on **P2**.1, Z step on **P2**.2... (same port!)

ej0rge, 2019-11-12:
Not according to Makerbase. 

https://github.com/makerbase-mks/MKS-SGen/blob/master/Hardware/MKS%20SGEN%20V1.0_001/MKS%20SGEN%20V1.0_001%20PIN.pdf

The marlin pins file for the SGEN starts with #include "../lpc1768/pins_MKS_SBASE.h", has some undefs for E1, and then defines for tmc220x uart pins. 

It's really confusing that SGEN-L is more of an SKR clone than a lesser SGEN. And it's not even really an SKR clone because the pins are very different on the SGEN-L than on the SKR. Can't imagine what possessed them to make a whole new design rather than just a new layout but there it is. 

All that being said, thanks for answering.

cprezzi, 2019-11-13:
Thanks for the link! It seems there is a pinout difference between SGen and SGEN_L.
According to their github the SGEN_L seems to have the mixed pinout:
https://github.com/makerbase-mks/Datasheet/blob/master/English%20datasheet
/MKS%20SGEN_L%20Datasheet.pdf

So at least the SGEN V1.0_001 should be working with grbl-LPC (without E1).