cprezzi/grbl-LPC #32: Z axis not homing with MKS

dansity, 2019-11-15:
Hello,
I have an issue where the Z axis is not homing using MKS board.
I have validated the cabling, the limit switch is recognized when pressed in Putty (`Pn:PZ`). X and Y axis homing without problem, Z not even moving. When I jog up manually the limit switch clicks but the Z does not stop.
It looks like Z axis is not considered at all. I have checked that GRBL is not in laser mode.
I have flashed the mks_sbase_firmware.bin from releases.
GRBL settings below:
$0 = 10    (Step pulse time, microseconds)
$1 = 255    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 4    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 19    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 200.000    (Homing locate feed rate, mm/min)
$25 = 800.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 2.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$33 = 5000.000   
$34 = 0.000   
$35 = 1.000   
$36 = 100.000   
$100 = 321.000    (X-axis travel resolution, step/mm)
$101 = 321.000    (Y-axis travel resolution, step/mm)
$102 = 320.000    (Z-axis travel resolution, step/mm)
$103 = 160.000   
$110 = 800.000    (X-axis maximum rate, mm/min)
$111 = 800.000    (Y-axis maximum rate, mm/min)
$112 = 400.000    (Z-axis maximum rate, mm/min)
$113 = 24000.000   
$120 = 2500.000    (X-axis acceleration, mm/sec^2)
$121 = 2500.000    (Y-axis acceleration, mm/sec^2)
$122 = 2500.000    (Z-axis acceleration, mm/sec^2)
$123 = 2500.000   
$130 = 800.000    (X-axis maximum travel, millimeters)
$131 = 800.000    (Y-axis maximum travel, millimeters)
$132 = 300.000    (Z-axis maximum travel, millimeters)
$133 = 100.000   
$140 = 0.600   
$141 = 0.600   
$142 = 0.000   
$143 = 0.000

cprezzi, 2019-11-15:
The precompiled version uses only X and Y homing (because it was made for a laser cutter).
If you need a different homing sequence, you would have to define it in config.h (or make an own machine profile in defaults.h) and compile your own binary.

dansity, 2019-11-15:
Do you have some little more in-depth description about compiling the files? I did find the lines to uncomment in config.h however i have no idea how to compile this thing and the readme mentions a very brief list of items to have for the compiling.

cprezzi, 2019-11-18:
Unfortunately I don't have any more detailed description as this:

- Install the `make` tool if not already installed (for Windows see http://gnuwin32.sourceforge.net/packages/make.htm)
- Install the ARM embeded toolchain (see https://developer.arm.com/open-source/gnu-toolchain/gnu-rm/downloads). If you use multiple toolchains on the same PC, you should use virtual machines.
- Include the path of `make` and the `arm-none-eabi-*` tools into your systems **path** constant.
- Clone this repo into a local folder.
- CD to this folder.
- Run `git submodule init` and `git submodule update`.
- Edit config.h, defaults.h, cpu-map.h in grbl folder to your needs.
- Run `make` to compile/build the binary. 
This produces 2 files:
  - build/firmware.bin: this is compatible with the sdcard bootloader.
  - build/grbl.hex: this is not compatible with the sdcard bootloader. It loads using Flash Magic and is primarilly for developers who don't want to keep swapping sdcards. If you flash this, then you'll have to reflash the bootloader if you want to go back!

dansity, 2019-11-18:
Thanks for your answer. I did managed to do this at the end however there was a ton of compiling error reference to the axis "A" in these files when i just simply cloned the repo and tried to build it.
I didn't really wanted to go into why that is happening so I just removed every single line from every file that mentioned axis "A" and then the `make` command worked out and now I have a file that is good for a 3axis CNC machine.
Thank you for your support once again.

cprezzi, 2019-11-18:
Are you sure these was errors not only warnings? I don't get any errors when I compile it.

dansity, 2019-11-18:
i got warnings for redefining the homing sequence but i got errors about 30-40 lines regarding every single A axis reference. I just removed them one-by-one and it compiled fine.
It could be my bad, I don't know. I don't know much about Make in general, this is our first met.

cprezzi, 2019-11-18:
Did you define the homing sequence in config.h? 
In my fork, this should not be defined in config.h, but in the corresponding machine profile in defualts.h.

dansity, 2019-11-18:
Well since I had no idea what I'm doing I did it in both location. :)

cprezzi, 2019-11-18:
The logic of my fork is, to have all supported board-profiles in cpu-map.h and machine-profiles in defaults.h. This way, I only have to `define` the board and machine in config.h and everything should be done.

dansity, 2019-11-18:
Ok cool. If i do anything next time I'll do like that. Thank you.
You can close the issue.

closed 2019-11-19 by cprezzi