cprezzi/grbl-LPC #38: MKS SGEN Version??

soko79, 2020-01-30:
Hi,

I would like to know if you are going to generate a version of your wonderful firmware for the MKS SGEN board.

Thanks and best regards

cprezzi, 2020-01-30:
According to the MKS info (see (https://github.com/makerbase-mks/Datasheet/blob/master/English%20datasheet/MKS%20SGen%20Datasheet.pdf), the step pins of the first 4 axes of the MKS Sgen V1.0 are on one port (same for dir pins), which means the first 4 axes could probably work for grbl-LPC. 

Unfortunately I don't have such a board for testing. If I get one for free, I will do my best to get it working.

cprezzi, 2020-01-30:
ATTENTION: The MKS SGEN_**L** is NOT compatible with grbl-LPC!

soko79, 2020-01-31:
OK, I only have a SGEN_L versión......If you ever program this version, I will be happy to try.

thanks

cprezzi, 2020-01-31:
As I said, grbl_LPC is NOT compatible with **SGEN-L**, because grbl needs to have all step pins on the same port and all dir pins on one port, which is not the case on the SGEN_L.

closed 2020-01-31 by cprezzi