cprezzi/grbl-LPC #39: Safety door detected as opened and door state initiated

betgear, 2020-02-12:
Latest release (11th Feb 2020). Just started to setup the mks-sbase v1.3 on the test bench.

Currently nothing connected (no steppers / endstops), when I connect with laserweb I get the error Error 13 - Safety door detected as opened.....

I've tried both grounding and applying 5v to p1.22 but the error does not clear.

Any help most appreciated.

cprezzi, 2020-02-13:
There was no change on the safety door code since last version. 
Could you please post your $$ settings?

betgear, 2020-02-13:
Hi,
It was a totally fresh install onto a new board, so presumably whatever the firmware sets as defaults. I will post them tonight when I get back from work.

betgear, 2020-02-13:
As requested

 Serial ports detected: [{"comName":"COM11","manufacturer":"Welch Allyn, Inc.","pnpId":"USB\\VID_FFFF&PID_0005\\DEADC0DE","locationId":"Port_#0004.Hub_#0003","vendorId":"FFFF","productId":"0005"}]
 Connecting Machine @ USB,COM11,115200baud
 Machine connected
 Firmware grbl 1.1f detected
 [MSG:'$H'|'$X' to unlock]
 Work Offset: 0 / 0 / 0 / 0
 Warning: Gcode out of machine bounds, can lead to running work halt
 Resetting alarm
 error: 13 - Safety door detected as opened and door state initiated.
 $$
 $0=10
 $1=255
 $2=0
 $3=3
 $4=0
 $5=1
 $6=0
 $10=0
 $11=0.010
 $12=0.002
 $13=0
 $20=1
 $21=0
 $22=1
 $23=1
 $24=50.000
 $25=6000.000
 $26=250
 $27=2.000
 $30=1000
 $31=0
 $32=1
 $33=5000.000
 $34=0.000
 $35=1.000
 $36=100.000
 $100=160.000
 $101=160.000
 $102=160.000
 $103=160.000
 $110=24000.000
 $111=24000.000
 $112=24000.000
 $113=24000.000
 $120=2500.000
 $121=2500.000
 $122=2500.000
 $123=2500.000
 $130=300.000
 $131=200.000
 $132=50.000
 $133=100.000
 $140=0.600
 $141=0.600
 $142=0.000
 $143=0.000

none007, 2020-02-13:
Hello, 
with Lightburn the same Problem with the new Release of Grbl-LPC

Waiting for connection...
Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
error:9
error:13
error:13
$$
$0=10
$1=255
$2=0
$3=3
$4=0
$5=1
$6=0
$10=0
$11=0.010
$12=0.002
$13=0
$20=1
$21=0
$22=1
$23=1
$24=50.000
$25=6000.000
$26=250
$27=2.000
$30=1000
$31=0
$32=1
$33=5000.000
$34=0.000
$35=1.000
$36=100.000
$100=160.000
$101=160.000
$102=160.000
$103=160.000
$110=24000.000
$111=24000.000
$112=24000.000
$113=24000.000
$120=2500.000
$121=2500.000
$122=2500.000
$123=2500.000
$130=300.000
$131=200.000
$132=50.000
$133=100.000
$140=0.600
$141=0.600
$142=0.000
$143=0.000
ok

cprezzi, 2020-02-13:
Ok, I need to check that with the board.

betgear, 2020-02-16:
I have been retro compiling old versions to see where the problem started.

It would appear this has been an issue since the door code was added on 5/7/18 commit. Versions before this worked fine (as the door code wasn't enabled).

The very first time the door code was enabled, this error occurs.

I'll do some more testing...

betgear, 2020-02-16:
OK, I think I've found the issue, the way the control_pin and control_mask are being used.

The original code assumes 8 bit ports, system.c : system_control_get_state().

The LPC is using a 32 bit port, and when reading bits 21 , 22 and 23 into the uint8_t the bits are being chopped off in the conversion from the 32 result to the 8 bit variable.

This could be happening elsewhere in the code as well if the original grbl code assumes 8 bit for port results.

The fix for this particular issue is to change the uint8_t to a uint32_t when reading the CONTROL_PIN in system_control_get_state().

Hope this helps.

betgear, 2020-02-16:
I've tested this now and it seems to work fine, for reference
P1.22 not connected to ground door closed.
P1.22 Connected to ground door open.

cprezzi, 2020-02-17:
Thank you for your help. I will change that in then code and recompile the bin files.

betgear, 2020-02-18:
I shall close the issue once the new versions are released and I've tested the released version :)

betgear, 2020-03-04:
Any news on when you're going to recompile the new versions without the door bug?

The release as it stands is broken and will not work on any hardware.

jaysuk, 2020-03-28:
Any update on this?
Betgear, would you consider releasing a temporary build or maybe even a pull request with this update?

betgear, 2020-03-29:
OK, I've done the pull request. Hopefully this is will implemented soon :)

cprezzi, 2020-03-30:
@betgear Thanks, I have merged your PR.

closed 2020-03-30 by cprezzi

jaysuk, 2020-03-30:
Will you compile and release new images?

cprezzi, 2020-03-30:
Yes, but it will take some time. I'm restricted to my home office for the next days/weeks and don't have access to my development environment here.

cprezzi, 2020-04-12:
I was able to compile all firmware versions today, so I close this issue now. Thanks for your help @betgear !