cprezzi/grbl-LPC #40: Re-Arm board update?

JeffHank1, 2020-02-28:
Thank you for porting this version of GRBL. I am still learning all of this... I have been using Grbl-Mega-5X on a Mega 2560 and want to upgrade to a Re-Arm board. Despite a lot of reading it is not clear to me which compiler to use for your version. Would the Arduino 1.8.8 IDE be suitable to make a bin file? I have Atom, Atmel, and Microsoft VS as well. I understand I have to rename a bin file firmware.bin and copy it to the SD card on the Re-Arm board, and reboot the board. I have tried this with the Smoothie bin version from your list but can't get my various Grbl capable communication programs to find Grbl-LPC on the board :( There is a jumper on the board to switch between USB and INT for power but the COM port will only be recognized when on USB. I am using a bare Re-Arm board for now. I would be happy to sort out the Re-Arm Pinout and compile a 5 axis CNC version. You were communicating with someone Sep 2019 with a similar question but I can't find any progress from there. Any suggestions are greatly appreciated. Thank you.

JeffHank1, 2020-02-29:
Oops. Kept reading and found the reference to Gnu Make and the alternate USB driver for Windows. I'll give those a go. Hopefully that driver won't screw up other USB board communicaitions. Cheers

cprezzi, 2020-03-30:
You don't need any IDE for compiling. Just install the tools like described in **README.md** under **Build notes**. Then compile the bin by running **make** on a dos prompt.

closed 2020-04-12 by cprezzi