cprezzi/grbl-LPC #41: Laser PSU wiring

Emiliogallo, 2020-03-08:
Hello, I have a diferent LPSU that you always coment, H L P G IN 5V.
This is mine
[image]
How is the correct wiring?
Thanks a lot.

Emiliogallo, 2020-03-09:
I have a MKS sBase 1.3

cprezzi, 2020-03-09:
Your LPS is the same as most have. Just connect the L, G and 24V of the right connector to the MKS board. L to PWM out, G and 24V to the power input for the mosfets and motors.

Emiliogallo, 2020-03-09:
Thanks for your fast reply.
Then it's just like this.
[image]
Any loop, or something else?

cprezzi, 2020-03-09:
Yes, but i'm not sure if the GND pin on the upper header of the MKS board is connected to the GND of the 12-24V input. You should checkt that.

Emiliogallo, 2020-03-09:
Ok, I'll check it. Thanks again.

Another question, please.
Is there any release with double y axis? Because I haven't a K40 Laser, it's DIY, and I have 2 motors on Y axis.

cprezzi, 2020-03-09:
No, GRBL does not support dual y-axis. 
But if you use external stepper drivers, you could connect two drivers to the step, dir, enagble pins of the y-axis.

Emiliogallo, 2020-03-10:
Thank you, cprezzi. I'll try everything this weekend.

Emiliogallo, 2020-03-14:
Wow, I'm really happy.
The above wiring diagram works great, just like you told me.
My engraves have improved very much, and all is faster now.
I've duplicate my Y axis just with a Cnc Shield and it's works propely too.
Thanks for all your support.

cprezzi, 2020-03-16:
You are welcome. I'm happy I could help.

closed 2020-03-16 by cprezzi