cprezzi/grbl-LPC #51: MKS SGEN (not the L version)

Wolfgang1us, 2020-11-30:
Hi,

I am trying for days to get the mks sgen board with MKS tmc2209 drivers up and running. I compiled 20 different versions based on the smoothieboard and the mks sbase, none of them is working. The binfiles are changed to .cur but the leds are not blinking the way they should. Disabled the power chip and also tried the c3d version. Connections from laserweb or lightburn are not possible. And other idea? The lastest Smoothieware installs fine and works...

Best regards, 
Wolfgang.

cprezzi, 2020-11-30:
Grbl-LPC is not compatiblewith the MKS SGEN board (as I already wrote many times)!
Grbl needs all direction pins on the same port and also all step pins on the same port, which is not the case on the SGEN.

theomajigga, 2020-12-07:
I run GRBL-LPC on the MKS SGEN L Board right now. I wish I could help you more.

closed 2021-02-19 by cprezzi