cprezzi/grbl-LPC #52: Dual axis homing

sigxcpu76, 2020-12-01:
Latest GRBL (1.1.h https://github.com/gnea/grbl/releases) implemented dual axis homing (with two motors & two endstops) to be able to square the gantry. Mostly used for dual Y motors, I think.
I tried to add that patch (https://github.com/gnea/grbl/commit/b75e5571eeaeb22a88304716fb1e7411f9c28be0) by hand, it compiled, but motors didn't move at all, so I don't know what I did wrong.
It is pretty clean to add it except the ISR handler, which is a bit different for LPC.

Do you think it may be easier for you to do it? This is a very useful feature for most mid-size CNCs that are using multiple motors.

The nice thing is that A axis is already there in the LPC port, so it can be used as second X or Y motor.

cprezzi, 2021-05-17:
Did you configure the stepper current $ settings? The default current is 0 (for security reason).

sigxcpu76, 2021-05-17:
I am not using the stepper drivers onboard. I use just the STEP,DIR,ENA pins with external steppers.

Nonstopich, 2021-12-08:
Good afternoon.
First of all, I want to say thank you for your work.

So your latest grbl-lpc release supports dual homming?
or second option: can I compile the latest sources to support dual homming functionality for lpc-based board?
My board is MKS SBase.