cprezzi/grbl-LPC #54: Smoothieboard 5XC-R

Wolfgang1us, 2020-12-08:
Hello,

after not beeing successful with the MKS SGen, I bought a 5XC-R. Smoothieware runs and all LEDs are blinking the way they should. As soon as I install the precompiled GRBL LPC Smoothieboard version, firmware.bin is renamed into FIRMWARE.CUR but only the first red led is on. Any idea?

Best  regards, 
Wolfgang.

cprezzi, 2020-12-08:
That's normal. Grbl doesn't use the other LEDs.

Wolfgang1us, 2020-12-09:
Good to known. Laserweb still disconnects after a couple of seconds , even with the .env file in place. Lightburn now finds the board, complaning about the safety door.

cprezzi, 2020-12-09:
It seems that LaserWeb sometimes doesn't read the .env file and therefore doesn't send a reset command after connecting. You can try to reset the board manually right after connecting vom LaserWeb.

Wolfgang1us, 2020-12-09:
Do you mean push the reset button on the board?

Not working:
Laserweb keeps disconnecting after a couple of seconds

Working so far:
All parameters are set using the Lightburn console. Now I got stuck with Error 13 Safety Door. No moves possible for further tryout.

Wolfgang1us, 2020-12-10:
Good morning, nobody able to help with error 13? Do I need to set a jumper?

cprezzi, 2020-12-10:
@Wolfgang1us Yes, I mean the reset button on the board.

There is no door switch configured in my precompiled firmwares. Did you activate it yourself?
Did you try to send $H first? Grbl goes to alarm state after boot / connect until you run homing.

Wolfgang1us, 2020-12-10:
@cprezzi
Thank you for your help. Just checked.
LW: As soon as I connect and press the reset button it looses the port and disconnects. 
LB: Finds Grbl 1.1f ok and error:13
Sending $H it writes 

Homing 
[VER:1.1f.20171008]
[OPT:VMD,249,8192]
Target buffer size found 
ok 
error:13

Nothing happens.

cprezzi, 2020-12-10:
I'm not sure that the `error:13` is realy comming from grbl. It could also be a error message of Lightburn. Did you check if this is documented on Lightburn?

cprezzi, 2020-12-10:
The usage of the feed hold pin for door switch is configured with a compile switch in config.h (https://github.com/cprezzi/grbl-LPC/blob/bbba8130b395de95a399d3ce31aaaccfaa339aff/grbl/config.h#L190). You can try to disable the door switch completely by remarking this line with a # in front.

Wolfgang1us, 2020-12-11:
@cprezzi Thank you very much. 
Commenting out the door pin solves error 13. Lightburn is now working.
I am still not able to connect to Laserweb...

cprezzi, 2021-02-19:
Please go to https://forum.makerforums.info/c/laserweb-cncweb/78 for help on LaserWeb.

closed 2021-02-19 by cprezzi