cprezzi/grbl-LPC #56: Is this still valid?

Primer-Merc, 2021-01-11:
I don't know where to ask this question, so please let me apologize if this is the wrong forum for questions.

I have a modified SBASE v1.3 with resistors R33,R34,R35,R39,R40,R41  removed. This was done to accommodate the use of PNP proximity switches.  ( Page 4 of this document if you need reference) https://github.com/makerbase-mks/MKS-SBASE/blob/master/Hardware/MKS%20SBASE%20V1.3_002/MKS%20SBASE%20V1.3_002%20SCH.pdf  

Switches are wired as normally closed. And pins are receiving 5V when not actuated. 
When configured with the following settings in Smoothie,  the limit switches act as expected.     

alpha_min_endstop                            1.24v!
alpha_max_endstop                           1.25v! 
beta_min_endstop                             1.27v!           
beta_max_endstop                            1.26v!
gamma_max_endstop                        1.29v!
gamma_homing_direction                 home_to_max

My question is..........   Is this note in the config.h still valid? 

//#define DISABLE_LIMIT_PIN_PULL_UP     Don't use; pull-up control not ported to ARM yet. pull-up is always on.  
//#define DISABLE_PROBE_PIN_PULL_UP     Don't use; pull-up control not ported to ARM yet. pull-up is always on.  
//#define DISABLE_CONTROL_PIN_PULL_UP   Don't use; pull-up control not ported to ARM yet. pull-up is always on. 

Will I be unable to to use grbl-LPC because I cannot tell the SOC to pull the pin to ground? 

Please forgive my ignorance if this seems rather easy. I'm old and still learning at a turtles pace.  My intention to change to grbl-LPC is solely based on the ability to have a live feed hold. Smoothie can't do feed holds with any reliability, and that's no good for a CNC router. 

Thanks,

Kyle

cprezzi, 2021-01-11:
Th best location to get help with this would probably be the maker forums: 
https://forum.makerforums.info/c/controllers/grbl-lpc/112

For your question: 
Yes, the software pull-up control was never ported (because most boards have hardware pull-ups, like SBASE v1.3 has).

Primer-Merc, 2021-01-11:
Thanks for the link cprezzi, I will definately post over there . I need pull down..if I understand correctly. Pulling down to ground.

closed 2021-01-11 by Primer-Merc