cprezzi/grbl-LPC #58: firmware.bin

Deltaplc, 2021-05-16:
Good morning
Sorry I am not very familiar with bin files because I have never had the need to create it. I play a little with arduino.
If I were to reconfigure grbl, for example by changing something on config.h ...... then how do I regenerate the firmware.bin file
with the changes I made?
I would like to modify a k40 laser and use an MKS SBase:

 https://www.amazon.it/3Dman-Controller-Compatible-Ramps1-4-Mega2560/dp/B07VRF4R3G/ref=pd_rhf_dp_p_img_4?_encoding=UTF8&refRID=HZQAS5HW3KKM6909MJ6E&th=1

Can it fit??
Grazie
Davide

cprezzi, 2021-05-17:
Please follow the **Build Notes** in README.md for compiling the bin file.
Hint: In grbl, most settings can be done with $ settings on runtime.

The board of your link (MKS **Sgen**) is not compatible with grblLPC. Only MKS **SBase** is.

closed 2021-05-17 by cprezzi

Deltaplc, 2021-05-17:
Grazie per la risposta.
Ti chiedo gentilmente se potresti mandarmi un link di esempio per quale  MKS SBase può essere compatibile con grbl?
Non vorrei comprare una scheda che non sia compatibile......
Questa può andare bene?
https://www.ebay.it/itm/363238430561?hash=item5492b29761:g:hJkAAOSw~kFf6wDU
Saluti
Davide

cprezzi, 2021-05-18:
Sorry, my italian is very bad. The link you posted is the correct board.

Deltaplc, 2021-05-19:
Thank you so much for your support and availability !!
Greetings
Davide