cprezzi/grbl-LPC #60: The current regulation of the motors is not working

qulet, 2022-03-27:
I compiled this version of GRBL 4 axes with, for the MKS-Sbase board. Broken current regulation $ 142, $ 143. It does not matter what the values are written, the Z and A axis motors are working and heating up a lot. $ 142 = 0,000
  $ 143 = 0,000
It does not disconnect the motors ... 

My configuration:
< $0=10          (Sets time length per step. Minimum 3usec.)
< $1=255         (Sets a short hold delay when stopping to let dynamics settle before disabling steppers. Value 255 keeps motors enabled with no delay.)
< $2=0           (Inverts the step signal. Set axis bit to invert (00000ZYX).)
< $3=5           (Inverts the direction signal. Set axis bit to invert (00000ZYX).)
< $4=0           (Inverts the stepper driver enable pin signal.)
< $5=1           (Inverts the all of the limit input pins.)
< $6=1           (Inverts the probe input pin signal.)
< $10=2          (Alters data included in status reports.)
< $11=0.010      (Sets how fast Grbl travels through consecutive motions. Lower value slows it down.)
< $12=0.002      (Sets the G2 and G3 arc tracing accuracy based on radial error. Beware: A very small value may effect performance.)
< $13=0          (Enables inch units when returning any position and rate value that is not a settings value.)
< $20=0          (Enables soft limits checks within machine travel and sets alarm when exceeded. Requires homing.)
< $21=1          (Enables hard limits. Immediately halts motion and throws an alarm when switch is triggered.)
< $22=1          (Enables homing cycle. Requires limit switches on all axes.)
< $23=3          (Homing searches for a switch in the positive direction. Set axis bit (00000ZYX) to search in negative direction.)
< $24=50.000     (Feed rate to slowly engage limit switch to determine its location accurately.)
< $25=6000.000   (Seek rate to quickly find the limit switch before the slower locating phase.)
< $26=250        (Sets a short delay between phases of homing cycle to let a switch debounce.)
< $27=2.000      (Retract distance after triggering switch to disengage it. Homing will fail if switch isn't cleared.)
< $30=1000       (Maximum spindle speed. Sets PWM to 100% duty cycle.)
< $31=0          (Minimum spindle speed. Sets PWM to 0.4% or lowest duty cycle.)
< $32=0          (Enables laser mode. Consecutive G1/2/3 commands will not halt when spindle speed is changed.)
< $33=5000.000   ( no information found '33')
< $34=0.000      ( no information found '34')
< $35=1.000      ( no information found '35')
< $36=100.000    ( no information found '36')
< $100=200.000   (X-axis travel resolution in steps per millimeter.)
< $101=200.000   (Y-axis travel resolution in steps per millimeter.)
< $102=800.000   (Z-axis travel resolution in steps per millimeter.)
< $103=18.000    (A-axis travel resolution in steps per degre.)
< $110=2400.000  (X-axis maximum rate. Used as G0 rapid rate.)
< $111=2400.000  (Y-axis maximum rate. Used as G0 rapid rate.)
< $112=2400.000  (Z-axis maximum rate. Used as G0 rapid rate.)
< $113=3000.000  (A-axis maximum rate. Used as G0 rapid rate.)
< $120=2500.000  (X-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $121=2500.000  (Y-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $122=2500.000  (Z-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $123=300.000   (A-axis acceleration. Used for motion planning to not exceed motor torque and lose steps.)
< $130=220.000   (Maximum X-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $131=280.000   (Maximum Y-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $132=200.000   (Maximum Z-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $133=360.000   (Maximum A-axis travel distance from homing switch. Determines valid machine space for soft-limits and homing search distances.)
< $140=0.600     ( no information found '140')
< $141=0.600     ( no information found '141')
< $142=0.000     ( no information found '142')
< $143=0.000     ( no information found '143')

cprezzi, 2022-03-28:
Can you please send a picture of your board? Unfortunately there are multiple versions sold under the name MKS SBase V1.3.

qulet, 2022-03-28:
[image]

qulet, 2022-03-28:
I have a Core xy 3d printer firmware is Merlin 2.09.01, Sometimes I would like to upload from the compiled GRBL-Hal 4 axes. Here's the problem
  Because the Z and A axis motors heat up very quickly even when the current is set to 0. In Marlin 👍 #define DIGIPOT_I2C_MOTOR_CURRENTS {0.9, 0.9, 0.9, 0.7, 0.7} and everything works "OK". pin 1.23. I have this PWM pin assigned to W Marlin.

qulet, 2022-03-28:
I have one more question regarding the Marlin build for this MKS SBASE board.
If I want to use the laser and RC servo of the Z axis probe in the firmware, it is not possible to compile. Is there any way around this problem?

cprezzi, 2022-03-29:
I'm confused. Are you using grbl-LPC, grblHAL or Marlin?
Unfortunately I don't know much about GRBL-Hal or Marlin. 

In grbl-LPC setting $140-143 to 0.0 will not disable the stepper drivers. It just sets the resistor value of the I2C digipot to to a minimum value. Did you try to set a value of 0.1 (for example)?

cprezzi, 2022-03-29:
If might have found the problem. Please try the following:
Change line 601 of the file grbl/cpu_map.h https://github.com/cprezzi/grbl-LPC/blob/6da1169f8da0f32aebe98835024d081b4fe72691/grbl/cpu_map.h#L601 to the following:
`#define CURRENT_WIPERS {0, 1, 2, 3};  // Wiper registers (X, Y, Z, E0)`

qulet, 2022-03-30:
367 / 5 000
Wyniki tłumaczenia
Setting $ 142 and $ 143 to 0.1 does not reduce the current supplied to the Z- and A-axis motors. MKS Sbase controls the 3d printer, Marlin is the basic software. Sometimes I want to replace the firmware with GRBL-HAL and use GRBL-PLOTTER software for the tangential knife.
I will change:
  #define CURRENT_WIPERS {0, 1, 2, 3}; // Wiper registers (X, Y, Z, E0)
Thank you for your answer

qulet, 2022-03-30:
Change:
  #define CURRENT_WIPERS {0, 1, 2, 3}; // Wiper registers (X, Y, Z, E0)
fixed the problem with motor overheating and current regulation. I have one more question :
What should be changed in the config.h file to get PWM for the diode laser to PIN1.23 (J8)
Thank you for your time !!!

cprezzi, 2022-03-30:
To compile a version with PWM on Pin 1.23 (instead of Port 2.5) you just have to activate line 42 (remove the //) in config.h:
https://github.com/cprezzi/grbl-LPC/blob/6da1169f8da0f32aebe98835024d081b4fe72691/grbl/config.h#L42

The direction port invert mask is a bitmask, where each bit corresponds to one axis in reverse order: 0000AZYX. So 00001101 = 13.

qulet, 2022-03-31:
[config.zip](https://github.com/cprezzi/grbl-LPC/files/8386954/config.zip)
There is no line #define SPINDLE_PWM_PIN_1_23 in config.h
The A Axis Mask Works!

cprezzi, 2022-03-31:
Just click this [Link](https://github.com/cprezzi/grbl-LPC/blob/6da1169f8da0f32aebe98835024d081b4fe72691/grbl/config.h#L42) and you will see the line I mean (which is gray).

qulet, 2022-03-31:
In this repository:
from the links from the site:
https://forum.makerforums.info/t/compiling-grbl-lpc-on-windows-im-starting-to-test-out-grbl-lpc-and-wanted-to-compile/7903

"If you want to download the 4 axis branch version
git clone -b more-axis https://github.com/cprezzi/grbl-LPC.git "
there is another config.h

git clone https://github.com/cprezzi/grbl-LPC
  there is the correct config.h file
  after compiling everything works "OK"
Thank you !

cprezzi, 2022-03-31:
Ok, that forum post was in 2017. Shortly after that, the 4th axis support was merged to the master branch and support fpr Pin 1.23 on MKS SBase was added later into the master branch. I probably should delete the more-axis branch...

closed 2022-03-31 by cprezzi