cprezzi/grbl-LPC #61: MKS Sbase  Current control  Z axis ,A axis not work

qulet, 2022-04-07:
Z and A axis motors after about 5 minutes of work on the following settings:
  $ 142 = 0.1
$ 143 = 0.1
they get very hot !!!

I changed my cpu_map for MKS Sbase according to the previous suggestion like this:

// Stepper current control
   #define CURRENT_I2C Driver_I2C1 // I2C driver for current control. Comment out to disable (for C3d boards!)
   #define CURRENT_MCP44XX_ADDR 0b0101100 // Address of MCP44XX
   #define CURRENT_WIPERS {0, 1, 2, 3}; // Wiper registers (X, Y, Z, E0)
   #define CURRENT_FACTOR 113.33 // Convert amps to digipot value
Unfortunately, this did not solve the problem with too much current fed to the motors.
The MCP44XX system itself is functional because the engines run on the Marlin 2.0 firmware

  #define DIGIPOT_I2C_MOTOR_CURRENTS {0.9, 0.9, 0.9, 0.4, 0.4} // MKS_SBASE

do not overheat.

cprezzi, 2022-04-07:
I am confused. In the other issue you wrote that the fix for CURRENT_WIPERS worked, and now you write it doesn't. What happend in between?

qulet, 2022-04-08:
I got the wrong impression that the fix worked, I checked it dry, that is, I didn't send g-gode. I had to quickly replace the firmware with Marlin and print something quickly. Yesterday I started sending the g-code to the tangential knife and two of my four motors after about 5 minutes got so hot that it was impossible to touch them ....

cprezzi, 2022-04-08:
I'm not sure if the schematics of the MKS SBase v1.3 I found are correct, but they show that Z and E0 current wipers are connected to ports 2 and 3 of the first digipot, but I'm not shure if that is correct. Most users of grbl-LPC are using it for laser cutters which only have 2 axes.

qulet, 2022-04-08:
[MKS SBASE V1.3_002.zip](https://github.com/cprezzi/grbl-LPC/files/8454605/MKS.SBASE.V1.3_002.zip)
Your GRBL-LPC branch was recommended as 4 axis.
Mksbase diagram in the appendix.

qulet, 2022-04-08:
[image]

cprezzi, 2022-04-09:
I used the same schematics according to which port 0 to 3 sould be mapped to x, y, z and e0. If that doesn't work, I might understand something wrong. For Smoothieboards, the values for the digipot mapping are 0, 1, 6, 7 (as I had it before also for MKS). 

Check out GRBL-HAL (https://github.com/grblHAL), which seems to also support MKS SBase.

qulet, 2022-04-10:
In the conig.h file I chose:
#define CPU_MAP_SMOOTHIEBOARD.
Smoothieboards, the values for the digipot mapping are 0, 1, 6, 7
  and compiled again - surprise ..
The engines stopped heating.

closed 2022-04-10 by qulet

cprezzi, 2022-04-11:
Ok, this means that the setting before was correct and there was another reason why they was not used. Maybe you did not activate CPU_MAP_MKS_SBASE before?

qulet, 2022-04-12:
Thank you for all your help in solving the problem!